Xinyuan-LilyGO/LilyGo-Cam-ESP32S3 #1: schematic of T-camera ESP32S3

lecrams, 2023-02-10:
Hi, I recently purchased the LilyGo ESP32 T-Camera S3 with OV2640 and pir. On the board it states V1.6. However checking the V1.6 schematic on https://github.com/Xinyuan-LilyGO/LilyGo-Camera-Series shows a wrover module while my board is equipped with a WROOM-1 module! According to the datasheets The pinning of the TROVER is different as the WROOM-1

Do you have a schematic available of the V1.6 HW with WROOM-1 module? That would allow me to develop my own code.

lewisxhe, 2023-02-10:
Which version did you buy? V1.6 ?

lecrams, 2023-02-10:
[image]
This board I did buy

lewisxhe, 2023-02-10:
oh I see. You need the schematic diagram. I thought I had uploaded the schematic diagram. Now you can see it

lecrams, 2023-02-10:
Thx. The title of the schematic say V1.5! 
Are there any differences with V1.6?

lecrams, 2023-02-10:
Is there also code (yaml) available for ESPhome?

lewisxhe, 2023-02-10:
The schematic diagram is the same
Yaml is not available at present because it still needs a power supply driver. As far as I know, there is no power supply driver in esphome

lecrams, 2023-02-10:
So this is an ESPhome issue? 
Are there no other lilygo modules with WROOM-1 that do have a yaml

lewisxhe, 2023-02-10:
So this is an ESPhome issue?
>Yes
Are there no other lilygo modules with WROOM-1 that do have a yaml
>At present, this version of esp32s3 is the only one without others

closed 2023-02-11 by lewisxhe