Xinyuan-LilyGO/LilyGo-Cam-ESP32S3 #2: can not build

lecrams, 2023-03-01:
I follow you guide how to build eg the allfunctions example I get the following error using VScode/platformio:

CMakeLists.txt not found in project directory d:\download\lilygo-cam-esp32s3-master\lilygo-cam-esp32s3-master.

did I miss something?

lewisxhe, 2023-03-01:
Why do you report cmake errors? Can you take a screenshot of your current editor's interface and errors

lecrams, 2023-03-01:
[image]
[image]

lecrams, 2023-03-01:
I do not see any ticked icon do I need to select another view as above?

lewisxhe, 2023-03-01:
[image]
In the lower left corner of the editor, isn't it?

lecrams, 2023-03-01:
I only see this at my bottom line:
[image]

lewisxhe, 2023-03-01:
It seems that your platformio plug-in is not activated. Please disable the espressf IDF plug-in in the plug-in, and then ensure that platformio is enabled. Then reopen vscode to see if platformio will be activated

lecrams, 2023-03-01:
that solved it. Thx for your support

closed 2023-03-01 by lewisxhe