Xinyuan-LilyGO/LilyGo-Cam-ESP32S3 #3: Streaming works only when serial monitor attached

maxpersg, 2023-03-11:
The source code seems to work only when a serial monitor is connected. If i disconnect the serial monitor, the stream stops and therefore i cannot connect this to a power plug to stream video. Any workarounds/

lewisxhe, 2023-03-11:
Add compilation parameters to `platfromioi.ini` so that it will not start USB printing and blocking
```
build_ flags =
-UARDUINO_ USB_ CDC_ ON_ BOOT
```

If you are using the arduino IDE, it is the same

Tools -> `USB CDC ON BOOT ` ->  Disable

maxpersg, 2023-03-12:
thank you. it worked. I'm trying to get this to work in ESPHome, but i keep getting this:

[W][esp32_camera:166]: Got invalid frame from camera!

lewisxhe, 2023-03-12:
ESPHOME is unable to use this board at present, because ESPHOME does not integrate the PMU driver code. The camera cannot work normally

maxpersg, 2023-03-12:
anyway if i can add this as a custom component to ESPhome?

lewisxhe, 2023-03-12:
Well, first of all, you need to turn on the camera voltage of the PMU to access the camera normally

maxpersg, 2023-03-12:
yes, but the comment i insert this line

    PMU.begin(Wire, AXP2101_SLAVE_ADDRESS, I2C_SDA, I2C_SCL);

i get this:
[I][i2c.arduino:175]: Performing I2C bus recovery 

it thn goes into an infinite reboot

lewisxhe, 2023-03-12:
'''PMU.begin(Wire, AXP2101_SLAVE_ADDRESS, 7,6);'''

Change to try this

lewisxhe, 2023-04-21:
still there ? If you still have problems, please reopen

closed 2023-04-21 by lewisxhe