Xinyuan-LilyGO/LilyGo-Cam-ESP32S3 #6: Problem with Esptool.py and cam_hal

richardgotthard, 2023-04-24:
Hi i'm kind of new to ESP32S3 but i occasionally get problems with flashing towards the camera using my ESP-IDF project. 

  File "/Users/richard/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/serial/serialutil.py", line 372, in timeout
    self._reconfigure_port()
  File "/Users/richard/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/serial/serialposix.py", line 401, in _reconfigure_port
    raise SerialException("Could not configure port: {}".format(msg))
serial.serialutil.SerialException: Could not configure port: (6, 'Device not configured')
CMake Error at run_serial_tool.cmake:56 (message):
  /Users/richard/.espressif/python_env/idf4.4_py3.10_env/bin/python
  /Users/richard/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip
  esp32s3 failed

FAILED: CMakeFiles/flash /Users/richard/esp/project/v2_sensor/build/CMakeFiles/flash 
cd /Users/richard/esp/esp-idf/components/esptool_py && /usr/local/Cellar/cmake/3.25.2/bin/cmake -D IDF_PATH="/Users/richard/esp/esp-idf" -D SERIAL_TOOL="/Users/richard/.espressif/python_env/idf4.4_py3.10_env/bin/python /Users/richard/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32s3" -D SERIAL_TOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="/Users/richard/esp/project/v2_sensor/build" -P /Users/richard/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.

I'm flashing using a USB-C cable but only sometimes after readjusting the cable it works. I've also tried using different cables as well as different LILYGO S3- devices. When it occasionally work i get connection problems with cam_hal : 

W (222581) cam_hal: Failed to get the frame on time!
Setting up first quick-shot... 
W (226881) cam_hal: Failed to get the frame on time!
Camera capture failed
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x420095b6  PS      : 0x00060230  A0      : 0x82009a61  A1      : 0x3fcf3b00  
0x420095b6: capture() at /Users/richard/esp/project/v2_sensor/main/main.cpp:156

A2      : 0x0000012c  A3      : 0x3fc997b0  A4      : 0x00000001  A5      : 0x00000000  
A6      : 0x3c0c67c8  A7      : 0x3fca6934  A8      : 0x820095b6  A9      : 0x3fcf3ae0  
A10     : 0x0000000a  A11     : 0x00036191  A12     : 0x3fc9e9fc  A13     : 0x00060023  
A14     : 0x00060020  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000c  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffa  

Backtrace: 0x420095b3:0x3fcf3b00 0x42009a5e:0x3fcf3b40 0x420b4380:0x3fcf3d40 0x403812e1:0x3fcf3d60
0x420095b3: capture() at /Users/richard/esp/project/v2_sensor/main/main.cpp:150

0x42009a5e: app_main at /Users/richard/esp/project/v2_sensor/main/main.cpp:420

0x420b4380: main_task at /Users/richard/esp/esp-idf/components/freertos/port/port_common.c:141 (discriminator 2)

0x403812e1: vPortTaskWrapper at /Users/richard/esp/esp-idf/components/freertos/port/xtensa/port.c:142

Does this have something to do with a cable i'm using or with the PMU in some way? Would appreciate any help possible.

VSCode: 1.77.3
ESP-IDF: v4.4.4
esptool.py v3.3.2

lewisxhe, 2023-04-24:
There are pin definitions here, you only need to configure the correct SDA, SCL, and the camera's Pin

https://github.com/Xinyuan-LilyGO/LilyGo-Cam-ESP32S3/blob/master/examples/AllFunction/utilities.h

richardgotthard, 2023-04-25:
I leave the pins to SDA, SCL and interupt as it is. But when i configure the file with power.h and power.cpp i get the following message when i boot up:

I (1400) sccb: sccb_i2c_port=1
I (1404) gpio: GPIO[39]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1443) camera: Detected camera at address=0x30
E (1444) sccb: SCCB_Write Failed addr:0x30, reg:0xff, data:0x01, ret:-1
I (1445) ov2640: Mismatch PID=0xff
E (1449) camera: Detected camera not supported.
E (1454) camera: Camera probe failed with error 0x106(ESP_ERR_NOT_SUPPORTED)
Camera init failed with error 0x106I (1462) FTP: connectFtpClient

I have the following configuration with power.cpp and power.h in the same project. I added the information from utilities.h inside my camera_pins.h, set configuration in setup.cpp and intialize "setuppower()" inside main together with a simple camera_init_function. 
[image]

richardgotthard, 2023-04-25:
I was able to correct this in the end by adding serial output-library in the project...
So there are no more issues regarding this! 

Thank you for the help!

closed 2023-04-25 by richardgotthard