Xinyuan-LilyGO/LilyGo-Cam-ESP32S3 #7: Problem installing MicroPython

mirrorleap, 2023-04-26:
Hi,

I tried to install MicroPython on the board with the Flash download tool using this guide - https://github.com/LilyGO/MicroPython-ESP32/blob/master/README.md. It seems to have flashed correctly.

But when I re-connected the board to my PC via USB, it started behaving strangely. It keeps on getting connected/disconnected continuously. I thought something has gone wrong with the flashing, so I tried again but the same thing happened.

Finally, I tried to flash the Factory Firmware - https://github.com/Xinyuan-LilyGO/LilyGo-Cam-ESP32S3/tree/master/firmware. Even though the flashing completed successfully, it keeps getting connected/disconnected when I connect it to the PC.

I'm new to ESP32 development, but could you please let me know what did I do wrong? And how I can flash MicroPython on the board correctly? I want to use Thonny or VSCode for programming.

Thank you.

lewisxhe, 2023-05-05:
The continuous disconnection is caused by flashing the wrong firmware and causing continuous restarts. At this time, you should follow the sixth operation of Tips in REAMDE

mirrorleap, 2023-05-06:
Thanks for the reply. I was able to fix this problem by flashing the factory firmware. However, I can't figure out the correct way to flash the MicroPython firmware.

Can you provide instructions on how to flash MicroPython correctly and which version? Thanks.

lewisxhe, 2023-05-07:
1. I did not use MicroPython, but to my knowledge, MicroPython does not support the programming method of ESP32 USB. Instead, it uses a USB TO TTL bridge for programming. If you need to write to microsython, you will need to use an additional USB TO TTL bridge for programming
Regarding firmware, most of LilyGo's boards use ESP32-R8 (OPI SPIRAM), which is currently not supported by MicroPython. You have two options,
1. Compile firmware yourself
2. Support for firmware using general ESP32S3 (without PSRAM)
3. The one you mentioned is ESP32, not ESP32S3, which is outdated [MicroPython-https://github.com/LilyGO/MicroPython-ESP32/blob/master/README.md](https://github.com/LilyGO/MicroPython-ESP32/blob/master/README.md)

mirrorleap, 2023-05-12:
Thanks for clarifying things and providing options to use MicroPython with ESP32-S3.

closed 2023-09-04 by mirrorleap