Xinyuan-LilyGO/LilyGo-Cam-ESP32S3 #9: PMU Power on by default

macf0x, 2024-01-03:
Hi Lewis,

I purchased this board a while back and noticed the ESPHome example provided.

I've not had much experience with the PMU on this board, but did read through the data sheet and your XPowers library looking for an answer.

Is there a way to configure the PMU to power on by default (via JST-GH), as occurs when powered by USB?

Or is it better to power via other means?

Regards
Rob

lewisxhe, 2024-01-05:
Hi,
AXP2101 cannot do this

macf0x, 2024-01-05:
Thanks Lewis,
I concluded as much, but good to get a definitive answer from someone familiar.
It's a shame there is no pad for VBUS. I guess USB is the only option.

closed 2024-01-05 by macf0x

lewisxhe, 2024-01-05:
[image]
Maybe you can fly a line to VBUS

macf0x, 2024-01-05:
I probed around, but didn't find an equivalent on the T-Camera S3 v1.6. It might be under the PIR.

Thanks for the suggestion