Xinyuan-LilyGO/LilyGo-Cam-ESP32S3 #11: Deep Sleep Current

brettbeeson, 2024-01-16:
Hi there Lewis,

I'd like to use this to deepsleep, wake on PIR, take a photo, deepsleep (and repeat).  This will be on **battery.**

I've previously tried this with a "esp32-cam" but we're unable to low current ([link](https://brettbeeson.com.au/mini-battery-and-solar-powered-timelapse-camera/))

Do you know the deepsleep current draw on this unit?

Thanks from Australia, Brett

lewisxhe, 2024-01-16:
https://github.com/Xinyuan-LilyGO/LilyGo-Cam-ESP32S3/blob/master/examples%2FAllFunction%2Fbutton.cpp#L60-L60

Here are the records of my testing

brettbeeson, 2024-01-16:
Thanks. That (400uA) is excellent for my purposes. That's impressively low - I guess the camera and screen are able to be turned off? 

Can you confirm my understanding (expect roughly 400uA in deep sleep with timer) is correct?

https://github.com/Xinyuan-LilyGO/LilyGo-Cam-ESP32S3/blob/master/examples%2FAllFunction%2Fbutton.cpp#L60-L60

lewisxhe, 2024-01-16:
yes

brettbeeson, 2024-01-18:
cheers, closing (and buying one)

closed 2024-01-18 by brettbeeson