Xinyuan-LilyGO/LilyGo-Cam-ESP32S3 #13: T-Camera S3 no longer works after programming with Arduino IDE

ralphjy, 2024-03-12:
I recently bought an enclosed T-Camera S3 on Amazon (with shell). It appeared to be programmed with the MinimalCameraExample in access point mode and it appeared to be woking properly. I used the setup described with Arduino IDE 2.3.2 to try to program the MinimalCameraStationExample. It compiled and loaded and successfully connected with my WiFi router, but then failed due to a Partition error.

`Connecting Wifi...

WiFi connected
IP address:
10.0.0.56
�[0;31mE (12576) MFN: Partition Not found�[0m
�[0;31mE (12577) MFN: Please Set the Partition�[0m`

After it fails, I can no longer upload changes to the device.

`Writing at 0x000eb997... (37 %)

A serial exception error occurred: Cannot configure port, something went wrong. Original message: OSError(22, 'A device which does not exist was specified.', None, 433)
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 1`

What is the partition issue and how do I recover?

Here is my configuration:
[image]

lewisxhe, 2024-03-13:
README FAQ 11

- Connect the board via the USB cable
- Press and hold the BOOT button , While still pressing the BOOT button, press RST
- Release the RST
- Release the BOOT button
- Upload sketch

ralphjy, 2024-03-13:
A couple of questions:
1) The left button on the case is marked RST, but doesn't seem to function as such
2) The right button on the case is marked A, but appears to also function as the BOOT button - do I need to open the case to access reset?

Also, how do I fix the partition problem?

closed 2024-08-31 by lewisxhe