Xinyuan-LilyGO/LilyGo-Cam-ESP32S3 #15: Send mail function

boubaj, 2024-08-28:
Hi,

I have tried to add the library ESP_Mail_Client.h ([mobizt/ESP-Mail-Client](https://github.com/mobizt/ESP-Mail-Client)) in the AllFunction example. But it does not work, either it stops and reboot (with SPIFFS) or it generates an error (partition spiffs not mounted) when I use LittleFS.

Can you tell me if it's possible to add a library that can send a mail with the example "AllFunction"?
I would like to send a basic mail when it triggers an event from the PIR.

KR,
Steve,

lewisxhe, 2024-08-31:
You are using platformio, right? Delete the following two lines, disable voice wake-up, and you can use the default spiffs partition

https://github.com/Xinyuan-LilyGO/LilyGo-Cam-ESP32S3/blob/33d3a67889664d806de3e4bee7ee4a5786ae08ee/platformio.ini#L19

https://github.com/Xinyuan-LilyGO/LilyGo-Cam-ESP32S3/blob/33d3a67889664d806de3e4bee7ee4a5786ae08ee/platformio.ini#L37

boubaj, 2024-09-02:
Hi,

Thank you very much.

KR,
Steve,

closed 2024-09-02 by boubaj