Xinyuan-LilyGO/LilyGo-Cam-ESP32S3 #16: My board won’t power on with regular 5V wall charger

gondhijagapathi, 2024-09-29:
[image]

it works with USB C though. I am new to this and confused how to get it to work

I am using ESP home and using yaml given in this repo 

https://github.com/Xinyuan-LilyGO/LilyGo-Camera-Series/blob/master/esphome/t-camera-s3.yaml

lewisxhe, 2024-10-01:
The external 5 Pin expansion socket of the board, 5V is shared with PMU SYS, please do not connect an external power supply load greater than 600mA, 3.3V use PMU DCDC3 power supply, the voltage can be adjusted, The maximum output current should not exceed 1A

gondhijagapathi, 2024-10-01:
@lewisxhe How do we power this board with an external power supply?

lewisxhe, 2024-10-04:
Use 4.2V to connect to the JST battery connector, or solder the wire to the position shown below. This is the USB 5v input capacitor, which is the same line as the USB 5V.
[image]

biospank, 2025-02-21:
@lewisxhe How do we power this board using usb connector and external power supply?
It works using usb cable conneted to my pc but it doesn't work using the same usb cable connected to an external power supply (5.0V - 1.0A).
Does it need more power?

Thank you

nnovofastovskiy, 2025-05-23:
Hi, I have a problem with VSYS. The voltage at the PMU_VSYS output when powered by the battery is not 5V but equal to the battery voltage (3.1V - 4.2V)

lewisxhe, 2025-05-23:
VSYS is not 5V, VSYS is equal to the input voltage

lewisxhe, 2025-05-23:
Oh! Sorry, this has not been changed. It is not a boost output. This is a path power supply, which is equal to the input voltage.

nnovofastovskiy, 2025-05-23:
> Oh! Sorry, this has not been changed. It is not a boost output. This is a path power supply, which is equal to the input voltage.

It's a pity to hear this, it turns out that when powered by a battery, the supply voltage of DCDC1 will be lower than 3.3V when the battery is discharged below 3.3V

lewisxhe, 2025-05-23:
Yes .

closed 2026-10-16 by github-actions[bot]