Xinyuan-LilyGO/LilyGo-Cam-ESP32S3 #17: E (472) cam_hal: cam_dma_config(301): frame buffer malloc failed E (478) cam_hal: cam_config(390): cam_dma_config failed E (484) gdma: gdma_disconnect(234): no peripheral is connected to the channel E (492) camera: Camera config failed with error 0xffffffff E (498) example:take_picture: Camera Init Failed

nitinGangurde, 2024-11-19:
/**
 * This example takes a picture every 5s and print its size on serial monitor.
 */

// =============================== SETUP ======================================

// 1. Board setup (Uncomment):
// #define BOARD_WROVER_KIT
// #define BOARD_ESP32CAM_AITHINKER
// #define BOARD_ESP32S3_WROOM

/**
 * 2. Kconfig setup
 *
 * If you have a Kconfig file, copy the content from
 *  https://github.com/espressif/esp32-camera/blob/master/Kconfig into it.
 * In case you haven't, copy and paste this Kconfig file inside the src directory.
 * This Kconfig file has definitions that allows more control over the camera and
 * how it will be initialized.
 */

/**
 * 3. Enable PSRAM on sdkconfig:
 *
 * CONFIG_ESP32_SPIRAM_SUPPORT=y
 *
 * More info on
 * https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-esp32-spiram-support
 */

// ================================ CODE ======================================

#include <esp_log.h>
#include <esp_system.h>
#include <nvs_flash.h>
#include <sys/param.h>
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

// support IDF 5.x
#ifndef portTICK_RATE_MS
#define portTICK_RATE_MS portTICK_PERIOD_MS
#endif

#include "esp_camera.h"

#define BOARD_ESP32S3_WROOM 1

// WROVER-KIT PIN Map
#ifdef BOARD_WROVER_KIT

#define CAM_PIN_PWDN -1  //power down is not used
#define CAM_PIN_RESET -1 //software reset will be performed
#define CAM_PIN_XCLK 21
#define CAM_PIN_SIOD 26
#define CAM_PIN_SIOC 27

#define CAM_PIN_D7 35
#define CAM_PIN_D6 34
#define CAM_PIN_D5 39
#define CAM_PIN_D4 36
#define CAM_PIN_D3 19
#define CAM_PIN_D2 18
#define CAM_PIN_D1 5
#define CAM_PIN_D0 4
#define CAM_PIN_VSYNC 25
#define CAM_PIN_HREF 23
#define CAM_PIN_PCLK 22

#endif

// ESP32Cam (AiThinker) PIN Map
#ifdef BOARD_ESP32CAM_AITHINKER

#define CAM_PIN_PWDN 32
#define CAM_PIN_RESET -1 //software reset will be performed
#define CAM_PIN_XCLK 0
#define CAM_PIN_SIOD 26
#define CAM_PIN_SIOC 27

#define CAM_PIN_D7 35
#define CAM_PIN_D6 34
#define CAM_PIN_D5 39
#define CAM_PIN_D4 36
#define CAM_PIN_D3 21
#define CAM_PIN_D2 19
#define CAM_PIN_D1 18
#define CAM_PIN_D0 5
#define CAM_PIN_VSYNC 25
#define CAM_PIN_HREF 23
#define CAM_PIN_PCLK 22

#endif
// ESP32S3 (WROOM) PIN Map
#ifdef BOARD_ESP32S3_WROOM
#define CAM_PIN_PWDN -1  //power down is not used
#define CAM_PIN_RESET 39 //software reset will be performed
#define CAM_PIN_XCLK 38
#define CAM_PIN_SIOD 5
#define CAM_PIN_SIOC 4

#define CAM_PIN_D7 9
#define CAM_PIN_D6 10
#define CAM_PIN_D5 11
#define CAM_PIN_D4 13
#define CAM_PIN_D3 21
#define CAM_PIN_D2 48
#define CAM_PIN_D1 47
#define CAM_PIN_D0 14
#define CAM_PIN_VSYNC 8
#define CAM_PIN_HREF 18
#define CAM_PIN_PCLK 12
#endif
static const char *TAG = "example:take_picture";

#if ESP_CAMERA_SUPPORTED
static camera_config_t camera_config = {
    .pin_pwdn = CAM_PIN_PWDN,
    .pin_reset = CAM_PIN_RESET,
    .pin_xclk = CAM_PIN_XCLK,
    .pin_sccb_sda = CAM_PIN_SIOD,
    .pin_sccb_scl = CAM_PIN_SIOC,

    .pin_d7 = CAM_PIN_D7,
    .pin_d6 = CAM_PIN_D6,
    .pin_d5 = CAM_PIN_D5,
    .pin_d4 = CAM_PIN_D4,
    .pin_d3 = CAM_PIN_D3,
    .pin_d2 = CAM_PIN_D2,
    .pin_d1 = CAM_PIN_D1,
    .pin_d0 = CAM_PIN_D0,
    .pin_vsync = CAM_PIN_VSYNC,
    .pin_href = CAM_PIN_HREF,
    .pin_pclk = CAM_PIN_PCLK,

    //XCLK 20MHz or 10MHz for OV2640 double FPS (Experimental)
    .xclk_freq_hz = 20000000,
    .ledc_timer = LEDC_TIMER_0,
    .ledc_channel = LEDC_CHANNEL_0,

    .pixel_format = PIXFORMAT_RGB565, //YUV422,GRAYSCALE,RGB565,JPEG
    .frame_size = FRAMESIZE_QVGA,    //QQVGA-UXGA, For ESP32, do not use sizes above QVGA when not JPEG. The performance of the ESP32-S series has improved a lot, but JPEG mode always gives better frame rates.

    .jpeg_quality = 12, //0-63, for OV series camera sensors, lower number means higher quality
    .fb_count = 1,       //When jpeg mode is used, if fb_count more than one, the driver will work in continuous mode.
    .fb_location = CAMERA_FB_IN_PSRAM,
    .grab_mode = CAMERA_GRAB_WHEN_EMPTY,
};

static esp_err_t init_camera(void)
{
    //initialize the camera
    esp_err_t err = esp_camera_init(&camera_config);
    if (err != ESP_OK)
    {
        ESP_LOGE(TAG, "Camera Init Failed");
        return err;
    }
    return ESP_OK;
}
#endif

void app_main(void)
{
#if ESP_CAMERA_SUPPORTED
    if(ESP_OK != init_camera()) {
        return;
    }

    while (1)
    {
        ESP_LOGI(TAG, "Taking picture...");
        camera_fb_t *pic = esp_camera_fb_get();

        // use pic->buf to access the image
        ESP_LOGI(TAG, "Picture taken! Its size was: %zu bytes", pic->len);
        esp_camera_fb_return(pic);

        vTaskDelay(5000 / portTICK_RATE_MS);
    }
#else
    ESP_LOGE(TAG, "Camera support is not available for this chip");
    return;
#endif
}
This code is giving error in esp-idf for lilgo t-camera s3 . Please solve the above code problem

lewisxhe, 2024-11-20:
You need to understand that this camera has a PMU and you must turn on the power to successfully initialize the camera.
https://github.com/Xinyuan-LilyGO/LilyGo-Cam-ESP32S3/blob/master/examples/MinimalCameraExample/MinimalCameraExample.ino

lewisxhe, 2024-11-21:
Please use Arduino code, or use compiled binary file to test the hardware.
https://github.com/Xinyuan-LilyGO/LilyGo-Cam-ESP32S3/tree/master/firmware

nitinGangurde, 2024-11-21:
I have tested the camera module with lilgo Cam- esp32s3 and it is working fine. But I want to use the esp-idf code to turn it ON. Please help me for this. I have merged the Power Library also to provide the power to the camera. But still it is giving me error.

nitinGangurde, 2024-11-21:
It is having in build PSRAM and I have configured it through my above code it is successfully allocating the PSRAM for it  e.g :- .fb_location = CAMERA_FB_IN_PSRAM. It is giving me error.

nitinGangurde, 2024-11-21:
I (26) boot: ESP-IDF HEAD-HASH-NOTFOUND 2nd stage bootloader
I (27) boot: compile time Nov 21 2024 11:12:37
I (27) boot: Multicore bootloader
I (31) boot: chip revision: v0.2
I (35) qio_mode: Enabling default flash chip QIO
I (40) boot.esp32s3: Boot SPI Speed : 80MHz
I (45) boot.esp32s3: SPI Mode       : QIO
I (49) boot.esp32s3: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00011000 00006000
I (78) boot:  1 phy_init         RF data          01 01 00017000 00001000
I (85) boot:  2 factory          factory app      00 00 00020000 00200000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=00020020 vaddr=3c030020 size=0c9c0h ( 51648) map
I (113) esp_image: segment 1: paddr=0002c9e8 vaddr=3fc93000 size=03630h ( 13872) load
I (116) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=223d8h (140248) map
I (144) esp_image: segment 3: paddr=00052400 vaddr=3fc96630 size=00238h (   568) load
I (144) esp_image: segment 4: paddr=00052640 vaddr=40374000 size=0ef84h ( 61316) load
I (167) boot: Loaded app from partition at offset 0x20000
I (168) boot: Disabling RNG early entropy source...
I (179) cpu_start: Multicore app
I (179) cpu_start: Pro cpu up.
I (179) cpu_start: Starting app cpu, entry point is 0x403753d4
0x403753d4: pthread_mutex_lock_internal at C:/Espressif/frameworks/esp-idf-v5.1.2/components/pthread/pthread.c:619

I (0) cpu_start: App cpu up.
I (197) cpu_start: Pro cpu start user code
I (197) cpu_start: cpu freq: 240000000 Hz
I (198) cpu_start: Application information:
I (200) cpu_start: Project name:     camera_example
I (206) cpu_start: App version:      v2.0.15-dirty
I (211) cpu_start: Compile time:     Nov 21 2024 11:15:11
I (218) cpu_start: ELF file SHA256:  a330b5d157eca3fc...
Warning: checksum mismatch between flashed and built applications. Checksum of built application is c7e9034ca755bb31384ac4fa90110d72a90678fdfed10b62d08ab41d55edc27f
I (223) cpu_start: ESP-IDF:          HEAD-HASH-NOTFOUND
I (229) cpu_start: Min chip rev:     v0.0
I (234) cpu_start: Max chip rev:     v0.99 
I (239) cpu_start: Chip rev:         v0.2
I (244) heap_init: Initializing. RAM available for dynamic allocation:
I (251) heap_init: At 3FC970A0 len 00052670 (329 KiB): DRAM
I (257) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (264) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (270) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (277) spi_flash: detected chip: gd
I (280) spi_flash: flash io: qio
W (284) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (298) sleep: Configure to isolate all GPIO pins in sleep state
I (304) sleep: Enable automatic switching of GPIO sleep configuration
I (312) app_start: Starting scheduler on CPU0
I32 m(I3 1(63)1 6a)p pm_saitna_rtta:s kS:t aSrttairntegd  socnh eCdPulU0er[ 0omn
 CPU1
I (321) main_task: Calling app_main()
I (326) s3 ll_cam: DMA Channel=0
I (329) cam_hal: cam init ok
I (333) sccb: pin_sda 5 pin_scl 4
I (337) sccb: sccb_i2c_port=1
I (340) gpio: GPIO[39]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (379) camera: Detected camera at address=0x30
I (382) camera: Detected OV2640 camera
I (382) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
I (460) s3 ll_cam: node_size: 3840, nodes_per_line: 1, lines_per_node: 6
I (460) s3 ll_cam: dma_half_buffer_min:  3840, dma_half_buffer: 15360, lines_per_half_buffer: 24, dma_buffer_size: 30720
I (467) cam_hal: buffer_size: 30720, half_buffer_size: 15360, node_buffer_size: 3840, node_cnt: 8, total_cnt: 10
I (478) cam_hal: Allocating 153600 Byte frame buffer in PSRAM
E (485) cam_hal: cam_dma_config(301): frame buffer malloc failed
E (491) cam_hal: cam_config(390): cam_dma_config failed
E (497) gdma: gdma_disconnect(234): no peripheral is connected to the channel
E (505) camera: Camera config failed with error 0xffffffff
E (511) example:take_picture: Camera Init Failed
I (516) main_task: Returned from app_main()

nitinGangurde, 2024-11-21:
I have provided the logs for my code

lewisxhe, 2024-11-21:
You did not enable PSRAM.
[image]

nitinGangurde, 2024-11-21:
When I enable the external PSRAM it is giving the error 
I (26) boot: ESP-IDF HEAD-HASH-NOTFOUND 2nd stage bootloader
I (26) boot: compile time Nov 18 2024 19:32:06
I (26) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (44) boot.esp32s3: SPI Flash Size : 16MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00200000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0fa70h ( 64112) map
I (111) esp_image: segment 1: paddr=0001fa98 vaddr=3fc93b00 size=00580h (  1408) load
I (112) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=294c8h (169160) map
I (147) esp_image: segment 3: paddr=000494f0 vaddr=3fc94080 size=0351ch ( 13596) load
I (151) esp_image: segment 4: paddr=0004ca14 vaddr=40374000 size=0fa4ch ( 64076) load
I (174) boot: Loaded app from partition at offset 0x10000
I (174) boot: Disabling RNG early entropy source...
I (185) cpu_start: Multicore app
E (186) quad_psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (189) cpu_start: Failed to init external RAM!

abort() was called at PC 0x403756a8 on core 0
0x403756a8: call_start_cpu0 at C:/Espressif/frameworks/esp-idf-v5.1.2/components/esp_system/port/cpu_start.c:474 (discriminator 3)

Backtrace: 0x40375a1e:0x3fceb230 0x4037b6ad:0x3fceb250 0x403818b6:0x3fceb270 0x403756a8:0x3fceb2e0 0x403cd9b4:0x3fceb340 0x403cddad:0x3fceb380 0x403c9971:0x3fceb4b0 0x40045c01:0x3fceb570 |<-CORRUPTED
0x40375a1e: panic_abort at C:/Espressif/frameworks/esp-idf-v5.1.2/components/esp_system/panic.c:452
0x4037b6ad: esp_system_abort at C:/Espressif/frameworks/esp-idf-v5.1.2/components/esp_system/port/esp_system_chip.c:84
0x403818b6: abort at C:/Espressif/frameworks/esp-idf-v5.1.2/components/newlib/abort.c:38
0x403756a8: call_start_cpu0 at C:/Espressif/frameworks/esp-idf-v5.1.2/components/esp_system/port/cpu_start.c:474 (discriminator 3)
0x40045c01: llc_task_random_gen_request in ROM

ELF file SHA256: e3bec33247757c30

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375d00
0x40375d00: esp_restart_noos at C:/Espressif/frameworks/esp-idf-v5.1.2/components/esp_system/port/soc/esp32s3/system_internal.c:158 (discriminator 1)
[image]

lewisxhe, 2024-11-21:
Which version of ESP-IDF are you using?

lewisxhe, 2024-11-21:
I watched 5.1.2, right?

nitinGangurde, 2024-11-21:
Yes

nitinGangurde, 2024-11-21:
Is their any bug in this version of esp-idf 5.1.2.

nitinGangurde, 2024-11-21:
I have solve the problem.

nitinGangurde, 2024-11-21:
Thanks for helping me and replaying me on this problem.

lewisxhe, 2024-11-21:
You need to set psram to octal mode, the default is qspi mode, right?

lewisxhe, 2024-11-21:
[image]

nitinGangurde, 2024-11-21:
No, need to set the PSRAM as DRAM at .fb_location = CAMERA_FB_IN_DRAM,

lewisxhe, 2024-11-21:
After opening psram normally, the buffer can be placed in psram

nitinGangurde, 2024-11-21:
I got confused for this After opening psram normally, the buffer can be placed in psram. I have just change the .fb_location = CAMERA_FB_IN_DRAM. And now it is working. Thanks for your help.

nitinGangurde, 2024-11-23:
While I am trying to get the frame of camera below warning pops up and it fails to get the frame.
W (96811) cam_hal: Failed to get the frame on time!
I (96811) APP_CODE_SCANNER: camera get failed

closed 2026-10-16 by github-actions[bot]