Xinyuan-LilyGO/LilyGo-Cam-ESP32S3 #18: readme is windows only

jonmartinsolaas, 2025-02-02:
After following the "instructions" the board was bricked just printing 

invalid header: 0x0a0a0a0a
invalid header: 0x0a0a0a0a
invalid header: 0x0a0a0a0a
invalid header: 0x0a0a0a0a
invalid header: 0x0a0a0a0a

must be something wrong with the image ... or the instructions.

lewisxhe, 2025-02-06:
Why do you say that? You can just select ESP32S3 DEV, then PSRAM DISABLE, select the default for Partition Scheme, and then just print the data to see if it is normal. You also need to set CDC to Enable.

lewisxhe, 2025-02-06:
Also, your problem is that you uploaded an invalid firmware, which caused the board to fail to boot. You can test the binary firmware to see if it is normal.https://github.com/Xinyuan-LilyGO/LilyGo-Cam-ESP32S3/tree/master/firmware

closed 2026-10-16 by github-actions[bot]