Xinyuan-LilyGO/LilyGo-Cam-ESP32S3 #21: t-camera-s3 as QR Code Reader

blondie63, 2025-05-20:
Hi, i'd like to use this board to read QR Code and send to a Rest Api server
I'm using PlatformIO that is very good and i can do all except manage a frame from camera and sed if there's a qrcode
I've tried to user some library, maybe this is the best: https://registry.platformio.org/libraries/alvarowolfx/ESP32QRCodeReader

The problem is that all test generate this error:
***ERROR*** A stack overflow in task loopTask has been detected.

Backtrace: 0x40377ea2:0x3fce8660 0x4037d401:0x3fce8680 0x40380852:0x3fce86a0 0x4037f054:0x3fce8720 0x4037d4b8:0x3fce8750 0x4037d4ae:0x00000000 |<-CORRUPTED

Please create an example project that you can then share here on github that simply does this:
In setup() activate the board and the Pir
In loop() wait for the Pir to detect someone in front and then activate the cam that takes the frames and decodes them to understand if a qrcode is shown, when it sees one it shows a message on Serial with the detected code

Thanks

lewisxhe, 2025-05-22:
Isn't this error a stack overflow? Have you tried changing the task stack size? What is your current stack size?

blondie63, 2025-05-22:
@lewisxhe The problem is not the stack size! i've tried also CONFIG_ARDUINO_LOOP_STACK_SIZE=32768
quirc lib for some reason corrupt memory, similar issue here: https://github.com/dlbeer/quirc/issues/131

lewisxhe, 2025-05-22:
The macro you changed is meaningless, because this macro is only valid when pre-compiled, and the subsequent definition has no effect. Now you need to confirm whether your stack overflow occurs inside the ESP32QRCodeReader library or in the user setup. Please post your code

blondie63, 2025-05-22:
@lewisxhe my source code here
[PIC-qr-main.zip](https://github.com/user-attachments/files/20391131/PIC-qr-main.zip)

lewisxhe, 2025-05-23:
[image]
Clearly this is an error within http, perhaps a null pointer was passed causing a kernel panic.

lewisxhe, 2025-05-23:
[image]
You did not check whether it was a null pointer, which caused the kernel to panic.

lewisxhe, 2025-05-23:
[image]
Finally, after solving the above problems, I encountered the stack overflow problem. As I mentioned above, it was obviously a problem of user task stack size overflow. It can be solved by simply creating an additional task and setting a large enough stack size.

lewisxhe, 2025-05-24:
Isn't there a running diagram above? ? ?
Didn't I say above that your setting is invalid? This macro doesn't work here at all

lewisxhe, 2025-05-26:
I am only discussing the stack overflow problem you encountered. I don't know what background you are talking about.
From the above picture, it can be seen that the board can run normally.

blondie63, 2025-05-26:
the board crash every time i try to use quirc lib !

lewisxhe, 2025-05-27:
I am running the code you provided, but you keep saying it doesn't work. Have you tested the code I modified?

blondie63, 2025-05-28:
@lewisxhe ok sorry, I see now that you put everything inside a task, now in fact it doesn't crash anymore even if it still can't decode my code, but that's another problem!
So you solved it by moving everything inside a Task?

lewisxhe, 2025-05-30:
emm... First of all, you need to know what the error is. It is because your memory has exceeded the size of the task stack configuration.

So you only need to create a new task stack that is large enough to solve this problem. As for your backend, I can't help you.

closed 2026-10-16 by github-actions[bot]